tetrahedron-technologies/pydhsfw #77: loopDHS not able to handle all incoming jpeg images

dsclassen, 2020-10-15:
I'm not sure why, but it appears that loopDHS (when running a collectLoopImages operation) is unable to keep up with the deluge of imaging coming into the jpeg receiver port. 

In this log snippet you can see that the first image that gets processed by AutoML and is returned to the DHS is `3`. Where did images `1` and `2` go?

```
2020-10-14 23:30:30,036 voltron.bl831.als.lbl.gov loopDHS.py-[dcss_start_operation():206] INFO FROM DCSS: stoh_start_operation collectLoopImages 1.2 0.03512688533520784
2020-10-14 23:30:30,036 voltron.bl831.als.lbl.gov loopDHS.py-[dcss_start_operation():209] INFO OPERATION: collectLoopImages, HANDLE: 1.2
2020-10-14 23:30:30,036 voltron.bl831.als.lbl.gov loopDHS.py-[collect_loop_images():254] INFO GOT: stoh_start_operation collectLoopImages 1.2 0.03512688533520784
2020-10-14 23:30:31,179 voltron.bl831.als.lbl.gov loopDHS.py-[automl_predict_response():402] INFO INDEX: 3 UL: [0.394978642, 0.537192941] LR: [0.540184915, 0.61789608]
2020-10-14 23:30:31,192 voltron.bl831.als.lbl.gov loopDHS.py-[drawBoundingBox():493] INFO WRITING: JPEGS/bboxes/automl_loop_0003.jpeg
2020-10-14 23:30:31,201 voltron.bl831.als.lbl.gov loopDHS.py-[automl_predict_response():407] INFO FOR DCSS: LOOP_INFO 3 normal 0.54018 0.04035 0.111 0.222 0.0807 0.39498 0.54018 0.53719 0.6179 0.14521 0
2020-10-14 23:30:31,227 voltron.bl831.als.lbl.gov loopDHS.py-[automl_predict_response():402] INFO INDEX: 6 UL: [0.394110024, 0.53744173] LR: [0.541780829, 0.617802]
2020-10-14 23:30:31,235 voltron.bl831.als.lbl.gov loopDHS.py-[drawBoundingBox():493] INFO WRITING: JPEGS/bboxes/automl_loop_0006.jpeg
2020-10-14 23:30:31,244 voltron.bl831.als.lbl.gov loopDHS.py-[automl_predict_response():407] INFO FOR DCSS: LOOP_INFO 6 normal 0.54178 0.04018 0.111 0.222 0.08036 0.39411 0.54178 0.53744 0.6178 0.14767 0
2020-10-14 23:30:31,284 voltron.bl831.als.lbl.gov loopDHS.py-[automl_predict_response():402] INFO INDEX: 7 UL: [0.395095, 0.539907873] LR: [0.540687084, 0.612865627]
2020-10-14 23:30:31,296 voltron.bl831.als.lbl.gov loopDHS.py-[drawBoundingBox():493] INFO WRITING: JPEGS/bboxes/automl_loop_0007.jpeg
2020-10-14 23:30:31,309 voltron.bl831.als.lbl.gov loopDHS.py-[automl_predict_response():407] INFO FOR DCSS: LOOP_INFO 7 normal 0.54069 0.03648 0.111 0.222 0.07296 0.39509 0.54069 0.53991 0.61287 0.14559 0
2020-10-14 23:30:31,334 voltron.bl831.als.lbl.gov loopDHS.py-[automl_predict_response():402] INFO INDEX: 9 UL: [0.397913277, 0.541441143] LR: [0.541903675, 0.611756146]
2020-10-14 23:30:31,345 voltron.bl831.als.lbl.gov loopDHS.py-[drawBoundingBox():493] INFO WRITING: JPEGS/bboxes/automl_loop_0009.jpeg
2020-10-14 23:30:31,356 voltron.bl831.als.lbl.gov loopDHS.py-[automl_predict_response():407] INFO FOR DCSS: LOOP_INFO 9 normal 0.5419 0.03516 0.111 0.222 0.07032 0.39791 0.5419 0.54144 0.61176 0.14399 0
2020-10-14 23:30:31,385 voltron.bl831.als.lbl.gov loopDHS.py-[automl_predict_response():402] INFO INDEX: 11 UL: [0.400690615, 0.542939723] LR: [0.540923953, 0.610094249]
2020-10-14 23:30:31,394 voltron.bl831.als.lbl.gov loopDHS.py-[drawBoundingBox():493] INFO WRITING: JPEGS/bboxes/automl_loop_0011.jpeg
2020-10-14 23:30:31,402 voltron.bl831.als.lbl.gov loopDHS.py-[automl_predict_response():407] INFO FOR DCSS: LOOP_INFO 11 normal 0.54092 0.03358 0.111 0.222 0.06715 0.40069 0.54092 0.54294 0.61009 0.14023 0
2020-10-14 23:30:31,424 voltron.bl831.als.lbl.gov loopDHS.py-[automl_predict_response():402] INFO INDEX: 12 UL: [0.404518664, 0.544202328] LR: [0.543596, 0.606935859]
```

It is also not uncommon to see this type of repeating pattern where the same image is being sent/received (see all the instances of `92` below)? Not sure where the problem lies so eyes on my code would be appreciated.

```
2020-10-14 23:30:34,044 voltron.bl831.als.lbl.gov loopDHS.py-[automl_predict_response():402] INFO INDEX: 90 UL: [0.382126451, 0.443920255] LR: [0.53952527, 0.561853051]
2020-10-14 23:30:34,053 voltron.bl831.als.lbl.gov loopDHS.py-[drawBoundingBox():493] INFO WRITING: JPEGS/bboxes/automl_loop_0090.jpeg
2020-10-14 23:30:34,066 voltron.bl831.als.lbl.gov loopDHS.py-[automl_predict_response():407] INFO FOR DCSS: LOOP_INFO 90 normal 0.53953 0.05897 0.111 0.222 0.11793 0.38213 0.53953 0.44392 0.56185 0.1574 0
2020-10-14 23:30:34,088 voltron.bl831.als.lbl.gov loopDHS.py-[automl_predict_response():402] INFO INDEX: 92 UL: [0.380781025, 0.435552061] LR: [0.536245584, 0.553778589]
2020-10-14 23:30:34,096 voltron.bl831.als.lbl.gov loopDHS.py-[drawBoundingBox():493] INFO WRITING: JPEGS/bboxes/automl_loop_0092.jpeg
2020-10-14 23:30:34,104 voltron.bl831.als.lbl.gov loopDHS.py-[automl_predict_response():407] INFO FOR DCSS: LOOP_INFO 92 normal 0.53625 0.05911 0.111 0.222 0.11823 0.38078 0.53625 0.43555 0.55378 0.15546 0
2020-10-14 23:30:34,123 voltron.bl831.als.lbl.gov loopDHS.py-[automl_predict_response():402] INFO INDEX: 92 UL: [0.381039709, 0.434716076] LR: [0.537179768, 0.552627921]
2020-10-14 23:30:34,132 voltron.bl831.als.lbl.gov loopDHS.py-[drawBoundingBox():493] INFO WRITING: JPEGS/bboxes/automl_loop_0092.jpeg
2020-10-14 23:30:34,140 voltron.bl831.als.lbl.gov loopDHS.py-[automl_predict_response():407] INFO FOR DCSS: LOOP_INFO 92 normal 0.53718 0.05896 0.111 0.222 0.11791 0.38104 0.53718 0.43472 0.55263 0.15614 0
2020-10-14 23:30:34,166 voltron.bl831.als.lbl.gov loopDHS.py-[automl_predict_response():402] INFO INDEX: 92 UL: [0.381408751, 0.43481338] LR: [0.536305308, 0.548030615]
2020-10-14 23:30:34,175 voltron.bl831.als.lbl.gov loopDHS.py-[drawBoundingBox():493] INFO WRITING: JPEGS/bboxes/automl_loop_0092.jpeg
2020-10-14 23:30:34,188 voltron.bl831.als.lbl.gov loopDHS.py-[automl_predict_response():407] INFO FOR DCSS: LOOP_INFO 92 normal 0.53631 0.05661 0.111 0.222 0.11322 0.38141 0.53631 0.43481 0.54803 0.1549 0
2020-10-14 23:30:34,207 voltron.bl831.als.lbl.gov loopDHS.py-[automl_predict_response():402] INFO INDEX: 92 UL: [0.37951, 0.434486598] LR: [0.537410796, 0.547031224]
2020-10-14 23:30:34,216 voltron.bl831.als.lbl.gov loopDHS.py-[drawBoundingBox():493] INFO WRITING: JPEGS/bboxes/automl_loop_0092.jpeg
```

gcmullen, 2020-10-15:
A few thoughts.. 
- Where is loopDHS running, on your machine at the house, or at the lab? 
- Our mini http server probably only allows one connection at a time, that means that the first image must be completely uploaded and disconnected before the next one can be sent. Anyway to get Axis logs to see if it's getting a timeout or a connection refused? 
- If this is the case, it would be nice to figure out how to get Axis to use the same connection to send all HTTP requests. Maybe an Axis server Software/Firmware update? Axis support call?
- There is a MultithreadMixin for our mini HTTPServer which *might* mean there is a thread pool that can handle multiple requests at a time... maybe 30. That *might* help out early on but we have have the same problem with the last half of the images because multiple connections doesn't mean they will upload any faster.

dsclassen, 2020-10-15:
The logs I posted above are with loopDHS running on Voltron at the beamline receiving images from the Axis server at 30fps (or as fast as possible up to 30fps).

I'm also NOT 100% certain that the problem is in the http code. The way that I store the images in the `LoopImageSet` class within the `jpeg_receiver_image_post_request` message handler and then use the `number_of_iamges` property as the "count" of images within the `automl_predict_response` message handler may be problematic.

gcmullen, 2020-10-15:
Try this... but don't save images to file for now, that really slow things down. You can enable again if it works. 
```
@register_message_handler('jpeg_receiver_image_post_request')
def axis_image_request(message:JpegReceiverImagePostRequestMessage, context:DhsContext):
    """
    This handler is triggered when a new jpeg image arrives from the jpeg_receiver. It is then shuttled off to AutoML.

    This may not be fast enough to keep up with 30fps coming from axis?
    """

    _logger.spam(message.file)
    # Store a set of images from the most recent collectLoopImages for subsequent analysis with reboxLoopImage
    if len(context.get_active_operations(operation_name='collectLoopImages')) > 0 and not context.gStopJpegStream:
        # add image to our jpeg list
        context.jpegs.add_image(message.file)
        # write to disk
        # saveJpeg(message.file)
        # Send to AutoMLPredictRequest message handler
        # generate unique key. 
        image_key = ''.join(choice(ascii_uppercase + digits) for i in range(12))
        context.get_connection('automl_conn').send(AutoMLPredictRequest(image_key, message.file))
    else:
        _logger.info(f'RECEVIED JPEG, BUT NOT DOING ANYTHING WITH IT.')
```

closed 2020-10-15 by dsclassen